Dronecode/camera-manager #13: Support video recording

anselmolsm, 2017-03-15:


rmackay9, 2017-05-07:
I'm also very keen on this item.

otaviobp, 2017-05-19:
Just to let it documented here, these MAVLink messages can be used in order to have video streaming support using MAVLink:
MAV_CMD_VIDEO_START_CAPTURE, MAV_CMD_VIDEO_STOP_CAPTURE, MAV_CMD_REQUEST_CAMERA_CAPTURE_STATUS

shakthi-prashanth-m, 2018-05-17:
I think https://github.com/Dronecode/camera-manager/pull/157 resolves this.

hamishwillee, 2018-05-18:
Yes, this can be closed. Streaming was supported for ages, and video saving/capture was delivered by #157. Controlled using MAVLink Camera API messages

Docs updated (really this is just the additional config in the [vidcap](https://hamishwillee.gitbooks.io/camera-manager/content/en/guide/configuration_file.html#vidcap) section of the config file.)

closed 2018-05-18 by hamishwillee