Dronecode/camera-manager #43: Add more detailed install instructions to README

rmackay9, 2017-05-15:
I'm having some problems getting this running.  I'm using an NVidia TX1 running Ubuntu but I plan to eventually run this on a few different "companion computers".

In any case, I'm having troubles at the "./autogen.sh && ./configure..." stage.  I get the message 

> 
> configure: error: Package requirements (avahi-client, avahi-core, avahi-glib) were not met:
> 
> No package 'avahi-client' found
> No package 'avahi-core' found
> No package 'avahi-glib' found
> 
> Consider adjusting the PKG_CONFIG_PATH environment variable if you installed software in a non-standard prefix

Sorry, I'm not much of a Linux/Ubuntu wizard.

rmackay9, 2017-05-15:
It seems like adding these packages helped so perhaps this line could be added to the README file:

> sudo apt-get install libavahi-client-dev libavahi-core-dev libavahi-glib-dev
> sudo apt-get install gstreamer-1.0
> sudo apt-get install libgstreamer-plugins-base1.0-dev
> sudo apt-get install libgstrtspserver-1.0-dev

rmackay9, 2017-05-15:
Also needed to install future to get "make" to work.

> sudo pip2 -q install -U future

lucasdemarchi, 2017-05-15:
Have you seeing the "Pre-requisites" section on the readme? Dependencies are listed there. If you can add there a subsection with needed packages for Ubuntu, it's ok.

otaviobp, 2017-05-15:
I would appreciate if you can update the README.md file with the information you gathered. I use this file to generate the build information in wiki page, so

otaviobp, 2017-05-15:
if you have that fixed, the wiki will also be updated.

rmackay9, 2017-05-16:
Sure, NP.  PR'd here: https://github.com/01org/camera-streaming-daemon/pull/46

closed 2017-05-16 by otaviobp