Dronecode/camera-manager #45: Reduce lag on video

rmackay9, 2017-05-15:
We've recently enabled the MP to be able to receive the video streams from csd.  It can be tested using this:

- install latest version of MP (http://ardupilot.org/copter/docs/common-install-mission-planner.html)
- ensure you're using beta version
   - Config/Tuning >> Planner >> Beta Updates checkbox
   - Help >> Check for BETA Updates
- Ensure the "companion computer" running csd and PC can reach each other on the network somehow
- On MP, select UDP and Connect
- Open Flight Data screen, press Ctrl-X.  A "Video" selection window should appear.  Select the format you'd like.  (this is quite beta so every available combination of format and resolution is displayed)

The issue is that the lag on the video stream is quite bad.  At least 2 seconds.  I suspect that the gstreamer pipeline isn't optimised.  In APSync's "simple video" system we use a pipeline like this:

gst-launch-1.0 -v v4l2src device=/dev/video0 ! video/x-raw,framerate=30/1 ! omxh264enc ! video/x-h264, stream-format=(string)byte-stream ! rtph264pay ! udpsink host=10.0.1.97 port=5600

There are probably better gstreamer pipelines but the above is at least an improvement I think.

lucasdemarchi, 2017-05-15:
Is this using Aero RTF? What camera?  I remember seeing data with it being around 200msec there.

The gstreamer pipeline you passed is very specific to one board (nvidia), so I guess you are using it? There's a way to tell csd what encoder you want to use. You need to add the "encoder" option in its configuration file.  @otaviobp can you point to an example configuration file with that set?

otaviobp, 2017-05-15:
That's true. To use the pipeline you described, you are going to need to create a config file in /etc/csd/main.conf with the following:
[gstreamer]
encoder=omxh264enc ! video/x-h264, stream-format=(string)byte-stream
muxer=rtph264pay
converter=video/x-raw,framerate=30/1

If you prefer you can created the file in other directory and run csd as:
./csd -c [your_config_file]

If you run csd with the -v option, all debug messages are going to be printed and you can check if the pipeline is as expected.

rmackay9, 2017-05-16:
great, thanks for the advice.  I'll give that a try.

I'm currently using an NVidia TX1 with a couple of Logitech C920 cameras because that's what I have sitting closest on my desk.  I think we're planning to use this multi-video stuff on a number of companion computer though including the RPi, TX1, Joule, etc.

fnoop, 2017-05-25:
@rmackay9 This is probably a client issue, not an issue at the streamer end.  Most rtsp clients seem to add a 2 second jitter buffer by default.  A good way to test is using gstreamer:
`gst-launch-1.0 rtspsrc location=rtsp://192.168.x.x:8554/video0 latency=0 ! decodebin ! autovideosink`
VLC almost makes it difficult to reduce latency.  In recent versions, open VLC preferences, go to 'Show All' (on Mac)/advanced.  In Input/Codecs (which is default when Show All opens for me), scroll down to Advanced and reduce Network Caching (ms) to 0.  In Input/Codecs->Demuxers->RTP/RTSP, change RTSP frame buffer size to 0.
In particular with gstreamer rtspsrc, the latency is excellent on a raspberry, <100ms.

lucasdemarchi, 2017-05-25:
@fnoop thanks for sharing. Is there any reason not to enable this by default? @otaviobp do you think this can be the default on the stream builder?

fnoop, 2017-05-25:
@lucasdemarchi Just to be clear, what I'm saying is that I don't think there's any problem at the server end - ie. with camera-streaming-daemon, the problem lies at the client end, ie. with MIssionPlanner/VLC.

lucasdemarchi, 2017-05-25:
Ohh... Sorry, now I see the gst launch is for the client side. Thanks.

closed 2017-08-17 by zehortigoza