Dronecode/camera-manager #67: Does Poky Aero (Intel Aero Linux Distro) 1.4.0-dev (pyro) contain all the dependencies?

cnpcshangbo, 2017-07-10:
From this page (https://github.com/01org/camera-streaming-daemon/wiki/Deploying-on-Aero), it is said we need to build an updated Aero image to run this daemon. I wonder if the image on https://download.01.org/aero/images/ is going to work.
I tried to run `scd` on Poky Aero (Intel Aero Linux Distro) 1.4.0-dev (pyro) and I got these message:
```
root@intel-aero:~# csd

(csd:723): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(csd:723): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(csd:723): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Unable to start rtsp server
AVAHI START
```
I wonder if I still need to build the image or I missed something to run `csd`?
Thanks,
Bo

anselmolsm, 2017-07-10:
Hi @cnpcshangbo 

1.4-rc2 already had the daemon, but I suggest you to try with 1.4 [we've just released](https://github.com/intel-aero/meta-intel-aero/releases/tag/v1.4). The image is on its way to become available for download from downloadcenter.intel.com.

csd daemon is enabled to be executed by systemd as part of the boot, so you should not need to run it manually.

closed 2017-07-10 by lucasdemarchi

cnpcshangbo, 2017-07-11:
We finally flashed to 1.4 version and the `csd` does run automatically as you mentioned. So this issue is solved. Thanks.