Dronecode/camera-manager #68: Where exactly should I run "make samples"?

cnpcshangbo, 2017-07-10:
I tried to run `make samples` on my desktop and I could not get a target.
Did I miss something?
[image]

anselmolsm, 2017-07-10:
Hi @cnpcshangbo . At the project's root directory (~/Downloads/camera-streaming-daemon-master in your case). ./autogen.sh c && make && make samples should configure, build csd and build the samples.

lucasdemarchi, 2017-07-10:
BTW, check the README.md file for additional build instructions.

closed 2017-07-10 by lucasdemarchi

cnpcshangbo, 2017-07-10:
Hi @anselmolsm . Thank you for you answer. I think I missed to follow the readme file.
So I found a desktop that has Ubuntu 14.04 and I tried to follow the readme file: (https://github.com/01org/camera-streaming-daemon/blob/master/README.md)
However, I got message like this:
```
bshang2@guoxiang-lab:~/Downloads/camera-streaming-daemon-master$ sudo apt-get install libgstrtspserver-1.0-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libgstrtspserver-1.0-dev
E: Couldn't find any package by regex 'libgstrtspserver-1.0-dev'
```
Then I rechecked the [Pre-Requisites](https://github.com/01org/camera-streaming-daemon#pre-requisites) and I found that part mentioned "GStreamer 1.4 or newer (https://gstreamer.freedesktop.org/)", however, [Installing Pre-Requisites on Ubuntu](https://github.com/01org/camera-streaming-daemon#installing-pre-requisites-on-ubuntu) part mentioned to install Gstreamer 1.0 (sudo apt-get install gstreamer-1.0).

So I wonder which version of Gstreamer should I install?

Am I doing the right method to compile or I need to do that on drones?

Thanks,
Bo

anselmolsm, 2017-07-10:
@cnpcshangbo The package name "gstreamer-1.0" is how distros differentiate from gstreamer-0.x versions.  For 14.04, it seems the packages available will install 1.2.4.

According to the [package search](https://packages.ubuntu.com/search?keywords=libgstrtspserver&searchon=names&suite=yakkety&section=all), libgstrtspserver is available in default repos since 16.10. For older Ubuntu releases, there might be extra repos, or you'll have to build by yourself.

> Am I doing the right method to compile or I need to do that on drones?
You should be able to build the samples this way, however notice 14.04 is pretty old and dependencies won't match.

cnpcshangbo, 2017-07-11:
@anselmolsm I have changed to Ubuntu 16 and the samples are compiled. We are able to run the `camera-sample-client`. 

[image]
However, when we use the vlc command to watch the video. Some error appears.
[image]
Do you have some ideas on the interface errors?
BTW, from the message `camera-sample-client` only shows RealSense cameras. Do we need to go to extended one to get videos from other cameras?
Thanks,
Bo