Dronecode/camera-manager #80: Streaming using Gstreamer/QGC not working with ov7251 VGA camera

RohanKurane, 2017-08-03:
Hello,

As suggested by @zehortigoza in https://github.com/intel-aero/meta-intel-aero/issues/223  , I am opening an issue to address the streaming issue with ov7251 VGA camera. 

I read frames using the camera and store it within my application. When the time comes to do streaming, I retrieve the frames one by one and use gstreamer/QGC to stream the video. I do the same for the realsense R200 camera and it works fine. Here is some of the code that builds the gstreamer pipeline and pushes the buffer to my ubuntu machine (192.168.8.2) where QGC is installed:

// create pipeline for ov7251, connect to 192.168.8.2 at port 5600 using UDP 
snprintf(str_pipeline, sizeof( str_pipeline ), "appsrc name=mysource ! videoconvert ! "
"video/x-raw,width=640,height=480,format=NV12 ! rtpjpegpay ! jpegenc ! videocrop top=0 left=0 right=40 bottom=10 ! " "udpsink host=%s port=5600", "192.168.8.2"); 

// launch the pipeline
pipeline = gst_parse_launch( str_pipeline, &error );

// set attributes
appsrc = gst_bin_get_by_name( GST_BIN( pipeline ), "mysource" );

app_caps = gst_caps_new_simple( "video/x-raw", "format", G_TYPE_STRING, "I420",
"width", G_TYPE_INT, WIDTH, "height", G_TYPE_INT, HEIGHT, NULL );

gst_app_src_set_caps( GST_APP_SRC( appsrc ), app_caps );

- Do the following for each frame stored in the application

// get the buffer and wrap it to send it over. frame_ptr and gst_size are the frame and frame sizes respectively.
gstbuffer = gst_buffer_new_wrapped_full( ( GstMemoryFlags )0, frame_ptr, gst_size, 0, gst_size, NULL, NULL ); 

//send buffer over to QGC to stream
g_signal_emit_by_name( appsrc, "push-buffer", gstbuffer, &ret );

Note: I am running on 1.3.1v and not 1.4v

192.168.8.2 is ip of ubuntu machine where QGC is installed

RohanKurane, 2017-08-08:
Is there any news on this ? 
@lucasdemarchi 
@zehortigoza

zehortigoza, 2017-08-17:
I just sent a PR(https://github.com/intel-aero/meta-intel-aero/pull/266) enabling the streaming of this camera in camera-streaming-daemon running in Intel Aero, so there is no issues with gstreamer lib of the 1.5.1v image, please compare both codes.

closed 2017-08-17 by zehortigoza