Dronecode/camera-manager #82: How to reduce latency when streaming video? [Avahi]

aho38, 2017-08-09:
I am using Intel Aero V1.4. When I stream video with vlc there's a latency of about 1.9 seconds. I wish to decrease that latency as much as possible. How can I change the resolution or what can I do?

zehortigoza, 2017-08-10:
Take a look in here: https://github.com/01org/camera-streaming-daemon/issues/45#issuecomment-303958941

aho38, 2017-08-11:
Thanks for that. One more problem I have is that although the latency was about 2 second, using vlc the video display is generally smooth. When I use gstreamer, it seems like the the latency is nearly 0; however, the video freezes every second or so. I tried following the instruction and changing latency for vlc in preferences, yet it does not change the actual display to the video. What can I do?

zehortigoza, 2017-08-11:
I also did not get any improvements with the VLC configuration, maybe in recent versions there is some other parameter to change.

About the gstreaming freezes this is the trade-off to have a lower latency. Over Wifi I also have some freezes but I belive that is mostly because there so many Wifi networks in my office.
Maybe you can try reduce the video resolution with:
```
gst-launch-1.0 rtspsrc location='rtsp://192.168.8.1:8554/video13?width=640&height=480' latency=0 ! decodebin ! autovideosink
```

closed 2017-08-17 by zehortigoza

fnoop, 2017-08-17:
@aho38 As @zehortigoza points out the point of the standard rtsp buffer is to reduce these streaming issues, but it doesn't mean that it's impossible to have low latency smooth streaming.  Possibly the main reason you'll see this is because the stream requires quite a high bitrate because csd hard codes the pipeline construction to compress and transmit each frame using jpg, rather than a video codec like h264.   So if you lower the resolution you will reduce the frame stream bitrate requirement and should see a lower rate of freeze/hangs.  In the future I'm sure csd can add support for more encoders.
VLC have changed where the stream buffer is set so you'll have to google your particular platform/version, but I managed to set it on both recent Mac and Linux versions to zero latency and it works well.
So in summary:
 - Maximise the quality of your wifi link.  Make sure your wifi connection bitrate supports at least the bitrate of the video stream
 - If you have problems with freeze/hangs, try reducing the resolution of the stream
 - CSD should add more encoder options, in particular h264/h265/vp[89] etc

Raghava248, 2019-05-22:
Iam facing delay around 10 seconds when trying to stream video using gstreamer and kinesis 
Iam using raspberry pi 3b
so please help me to solve the latency problem
This is the command iam using
[image]

MatejFranceskin, 2019-05-22:
To see which part is responsible for latency try to change the sink to udpsink host=v.x.y.z port=1234 and then try to play it on the other computer with gst-launch-1.0 udpsrc port=1234 ! "application/x-rtp, payload=127" ! rtph264depay ! avdec_h264 ! xvimagesink sync=false
If there is no delay there it is either kvssink or most likely the player or service that is doing buffering before the video is displayed.