Dronecode/camera-manager #85: Pixhawk Flight controller connections

RohanKurane, 2017-09-28:
Hi,

@zehortigoza 
@lucasdemarchi 

P.S : This is the wrong place for this question but I dont seem to find a way to open a new issue in PX4 section. Please move there accordingly. 

I have a Pixhawk FC that I bought. I have it powered thru a 4S lipo battery. I have the TELEM2 connected to the J5 Telemetry slot on the intel compute board (80 pin connector extension). The compute board is powered thru its HSUART connection. 

How do I get the QGC app that is running on my workstation to detect the Pixhawk ? I am trying to simulate a flying drone and view it on QGC. I have my application ported to the compute board and want to see MAVLINK messages in my application sent by the Pixhawk while viewing the drone on the QGC. 

I have found mixed documentation and none of them tell me how to get the QGC detect the Pixhawk. 
Can you please help me out here ? 
[image]

Thanks
Rohan

lucasdemarchi, 2017-09-28:
To connect to pixhawk you should use the wires blue, yellow and black coming out from the same connector where you powered aero.

Make sure you are using 5V to power Aero. The pins on Tha 80-pin connector should not be used. If you do want to use the 80-pin connector (for example because you don't have the cable) then you will need to change the fpga firmware to send the signal to the correct pins. Note that you have to remove the red wire as both boards are powered independently.

Then you should flash the fpga firmware that targets the compute-board, not the RTF (see in /etc/fpga)

lucasdemarchi, 2017-09-28:
For the connector pinout you can use https://www.intel.com/content/dam/support/us/en/documents/boardsandkits/aero/intel-aero-compute-board-guide.pdf

See section 3.8

And make sure you understand the "caution" notes there and that VCC and GND on the telemetry connector don't match the pixhawk connector.

zehortigoza, 2017-09-28:
All those ports(telemetry, compass...) are connected to FPGA and the RTF FPGA firmware routes it to AeroFC that you don't have, so like Lucas said you need to do another FPGA firmware(you can use https://github.com/intel-aero/intel-aero-fpga as a start point) to route the Compute board UART to the telemetry connector. It is easy just use the UART pins in the power connector of compute board.

RohanKurane, 2017-09-28:
Thanks @lucasdemarchi  and @zehortigoza. So if I understand this both of you correctly, 

- Power Pixhawk thru 4S lipo and power pinout on Pixhawk.

- Flashed the fpga firmware by running jam - aprogram /etc/fpga/aero-compute-board.jam

- I wont connect the TELEM2 pinout to the 80 pin connector. I will use the 3 pinout from the other end of the HSUART power cable used to power the compute board. Where on the pixhawk does the other go ? There are 16 rows of 3 pins each on the top of the Pixhawk. I assume the other end goes there. But which 3 pins do I connect it to ?  I am attaching a pic here. 
[image]

- So there is no need to separately connect the TELEM2 pins on the Pixhawk I assume ?

- How do I connect the Pixhawk to my workstation so QGC can detect it ? I see a micro USB connection on one side of the Pixhawk. 

Thanks
Rohan

zehortigoza, 2017-09-28:
You need to connect compute board to a telemetry port in Pixhawk, this way compute board can receive and send MAVLink messages to Pixhawk.

The easiest way to connect Pixhawk to your computer is by using the microUSB there is other ways too like connect a USB FTDI to you computer and wire the FTDI to a telemetry port in Pixhawk.

RohanKurane, 2017-09-28:
The 3 pinout from the HSUART does not fit in to the Telemetry pinout directly. I got some cable with the Pixhawk that would fit into the 3 pinout. I have attached 2 pics. 

First of the 3 pin out from the HSUART
[image]

Here is the cable that fits into the 3 pinout. 
[image]

Let me know if there is another cable that I dont have and should order.  The Telemetry is a 6 pin connection. 

If you have any pics that you can share that would be nice.

RohanKurane, 2017-09-28:
To reply to your comment @zehortigoza 
All those ports(telemetry, compass...) are connected to FPGA and the RTF FPGA firmware routes it to AeroFC that you don't have 
>> I do not have a RTF at the moment. I am using the fpga px4 firmware that came installed on the compute board. I assume that is good enough to work with Pixhawk.

It is easy just use the UART pins in the power connector of compute board. 
>> Please see my last comment with pics. How does that 3 pinout connect to the Telem2 on the Pixhawk. 

Also, when I connect the Pixhawk to the laptop using the USB connection, I get a red solid light for the main LED and the B/E for the IO is a solid orange

zehortigoza, 2017-09-28:
You can just cut and solder the cable that you have, no need to order. Again as Lucas said you need to load this FPGA firmware /etc/fpga/aero-compute-board.jam 
I don't know the meaning of this solid orange led but you can find the answer to that in the Pixhawk documentation.

RohanKurane, 2017-09-28:
@zehortigoza  
Again as Lucas said you need to load this FPGA firmware /etc/fpga/aero-compute-board.jam
>>I did flash the firmware you are mentioning. I ran
jam -aprogram /etc/fpga/aero-compute-board.jam

and got success.  I assume thats all I need to do there. 

You can just cut and solder the cable that you have, no need to order. 
>>I am confused. How is the 3 pinout suppose to connect originally ? What is the use of it ? Its got to fit somehow into the Pixhawk FC. When you say xcut and solder, what exactly ? That seems complicated for something that cost me $200 and I dont want to mess it up. 

I don't know the meaning of this solid orange led but you can find the answer to that in the Pixhawk documentation.
>> I googled it. Apparently a solid Amber B/E and a flashing blue ACT means error and everything OK at the same time. That is strange.  Here is what I found
The LEDs are
 FMU PWR = Flight management Unit Power
 FMU B/E = FMU Processor in Bootloader mode (flashing) or Error (Steady)

I/O PWR = Input/ Output Power
I/O B/E = I/O processor in Bootloader mode (flashing) or Error (Steady)
ACT = Activity (a flashing heartbeat) indicating everything is ok

zehortigoza, 2017-09-28:
You can connect this(https://user-images.githubusercontent.com/18452209/30987689-c5196514-a44c-11e7-8af4-177aba3e7eb1.jpg) to this(https://user-images.githubusercontent.com/18452209/30987695-c8a22770-a44c-11e7-8255-d23c1ebaa409.jpg) then cut the other side of this connector and connect to a cable with the connector that connects to the Pixhawk telemetry port.

You just need to make sure about the order, in the aero side the black in GND the other 2 is UART RX and UART TX somewhere in the Aero documentation(https://github.com/intel-aero/meta-intel-aero/wiki) you can find it for sure but it will not cause any damage connecting inverted.

RohanKurane, 2017-09-28:
@zehortigoza then cut the other side of this connector and connect to a cable with the connector that connects to the Pixhawk telemetry port.
>> I dont have any such cable and nothing came with the Pixhawk. I can see if anyone solders it. Can you please send me a picture of your connection if you have one ?  I understand Pixhawk is not specifically designed for intel aero but there must be a way to do this without soldering. I do not know what the RTF uses and I am not sure how the board and FC are connected inside of it. I need to find a cable that has a 3 in and 6 out connector.

zehortigoza, 2017-09-28:
I don't have a Pixhawk.

You can cut one side this cable that you are using in this pic https://user-images.githubusercontent.com/18452209/30983285-d90ba676-a43e-11e7-8e88-9eb36a798780.jpg then connect to compute board.
Look here: https://pixhawk.org/modules/pixhawk#pinouts you just need to connect 3 pins: GND, TX and RX. TX and RX are UART TX and RX.

zehortigoza, 2017-11-02:
This is not the right place for this issue and by the missing of answers looks like it is fixed.

closed 2017-11-02 by zehortigoza