Dronecode/camera-manager #88: Video Streaming randolmly stops using QGC and GStreamer

RohanKurane, 2017-12-11:
Sometimes while streaming the video from the RealSense camera and viewing on QGC, the video streaming randomly stops after few frames are read and sent to QGC using GStreamer. The QGC screen dosent show anything after but the camera is still recording (red light is ON)

Is this a known issue ? Is there anything I can do to debug why this may be happening ? I turned on GST_DEBUG but didnt see anything that would help.

lbegani, 2017-12-11:
> Sometimes while streaming the video from the RealSense camera and viewing on QGC, the video streaming randomly stops after few frames (sometimes 10, sometimes 50) are read and sent to QGC using GStreamer. The QGC screen dosent show anything after but the camera is still recording (red light is ON)

Can you check if drone running csd is still connected to QGC. It may happen due to connection loss.
> Does QGC expect a certain framerate and if it does not receive it, this could happen ? The framerate is set on the GST pipeline during get launch to 30fps.

No, I dont think so.

RohanKurane, 2017-12-11:
Good point. Today I tested again. Even before my video streaming starts, I noticed that the Aero wifi hotspot connection is lost. How can I fix this ? I have not noticed this before today. The wifi connection is lost.

RohanKurane, 2017-12-12:
I checked and the connection is not lost during video streaming. But no video appears after a while. How can i debug ?

lbegani, 2017-12-12:
Can you test with drone and QGC connected over USB instead of Wifi.  If reproduces, it will rule out the wifi connectivity issue.
You can also run the csd in ubuntu system with webcam and run the QGC in the same system to see if there is any issue.
You can also try running csd with logging enabled ("-v")

RohanKurane, 2017-12-12:
I do not believe this is a connectivity issue anymore. The connection persists. However no video shows after a few frames on QGC. The buffers are sent correctly because gst_app_src_push_buffer() returns GST_FLOW_OK. 
I check the pipeline state after every send and for the first frame it is PAUSED and thereafter it is PLAYING. 
But the video stops after a few frames (few can vary from 2 to 10). 
My pipeline is built as

char str_pipeline[200];

snprintf(str_pipeline, sizeof( str_pipeline ), "appsrc name=mysource ! videoconvert ! "
    "video/x-raw,width=640,height=480,format=NV12 ! vaapih264enc ! h264parse ! rtph264pay !"
    "udpsink host=%s port=5600", "192.168.8.2");  

pipeline    = gst_parse_launch( str_pipeline, &error );

 if( !pipeline )
   {
     printf( "gst_parse_launch error. Cannot launch GStreamer..: %s\n", error->message );
      return; 
   }

  appsrc = gst_bin_get_by_name( GST_BIN( pipeline ), "mysource" );

  app_caps = gst_caps_new_simple( "video/x-raw", "format", G_TYPE_STRING, "RGB",
                    "width", G_TYPE_INT, WIDTH, "height", G_TYPE_INT, HEIGHT, NULL );

gst_app_src_set_caps( GST_APP_SRC( appsrc ), app_caps );
gst_caps_unref( app_caps );
g_object_set( G_OBJECT( appsrc ), "is-live", TRUE, "format", GST_FORMAT_TIME, NULL );

Is there something in my pipeline that is wrong ?  What is a correct pipeline to stream RGB COLOR video ? Am I missing some pipeline elements ?

lbegani, 2017-12-13:
Do you see issue in video streaming with CSD or with your application?

RohanKurane, 2017-12-13:
I believe it might be the way the pipeline is constructed. I'm not sure. Do you have a reference pipeline for video streaming RGB frames ? Take a look at mine above. The pipeline remains in Paused state instead of changing to Playing state.

RohanKurane, 2017-12-13:
I could be missing some elements or have the wrong elements in my pipeline. I think.

lbegani, 2017-12-13:
As I now understand, issue is with your application and not with camera-streaming-daemon.
Try this pipeline for udp streaming and see if it works - 
`$ gst-launch-1.0 videotestsrc ! video/x-raw,width=640,height=480 ! videoconvert ! x264enc ! rtph264pay ! udpsink host=0.0.0.0 port=5600`
Check for the logs in the QGC side to see if its throwing any error regarding the format of the video stream.

RohanKurane, 2017-12-14:
The pipeline did not seem to help. I do not know how to view the logs on QGC, If I refresh the logs button in QGC, nothing appears. Do I need to set something on my UAV to see the logs on QGC ?

lbegani, 2017-12-14:
You can go through the below links for information that can help you -
1. https://dev.px4.io/en/qgc/video_streaming.html
2. https://dev.qgroundcontrol.com/en/

closed 2018-01-22 by RohanKurane