Dronecode/camera-manager #107: Document/Move Samples

hamishwillee, 2018-02-14:
There is already a [Samples](https://github.com/intel/camera-streaming-daemon/wiki/Samples) doc in the wiki. This should *possibly* be renamed as Exampes or Test code or Test tools - that is after all what they are.

In any case, lets move out of wiki and into same doc tree as other docs.

lbegani, 2018-03-06:
@SaliniVenate 
1. Move out sample code to <csd>/test/ directory
2. Rename file with test_<>.cpp
3. Compile and test the unit feature
4. Document steps to compile, test

hamishwillee, 2018-03-06:
FYI, I started to re-document the samples: https://hamishwillee.gitbooks.io/dronecode-camera-server-sdk/content/en/examples/samples.html

MOving most of them to test/ sounds excellent. Only concern is [camera-sample-custom](https://hamishwillee.gitbooks.io/dronecode-camera-server-sdk/content/en/examples/samples.html#camera-sample-custom-%E2%80%94-demo-code-for-extending-csd) since that appears to be more "demo code accompanying guide" for extending CSD than a test.

hamishwillee, 2018-03-22:
Sample build warning removed: https://github.com/intel/camera-streaming-daemon/issues/130

hamishwillee, 2018-03-22:
Closing as done. There is broader work to be covered in testing, but that is captured in other issues.

closed 2018-03-22 by hamishwillee