Dronecode/camera-manager #110: Document Extension Interfaces

hamishwillee, 2018-02-14:
Document Extension Interfaces:
 - Supporting new camera - e.g.:
   - how to support a new real camera (FLIR, RealSense, Sony)
   - how to support Gazebo camera/simulated camera
- Support new streams/stream types - some material in [Extending-Camera-Streaming-Daemon](https://github.com/intel/camera-streaming-daemon/wiki/Extending-Camera-Streaming-Daemon) (wiki) 
- Are there any other custom interfaces?

@lbegani This one for you.

lbegani, 2018-02-17:
How to extend to support -

1. New real camera device
2. New Gazebo simulated camera
3. New custom (user-defined) parameters by camera device

hamishwillee, 2018-03-21:
This will be addressed by https://github.com/Dronecode/csd_docs/pull/8

closed 2018-03-21 by hamishwillee