Dronecode/camera-manager #115: Avahi is a non-optional component

hamishwillee, 2018-02-26:
Avahi is required to generate the list of available RTSP streams. This is supposed to be an optional feature and component.

However following the default instructions seem to imply it is required.
```
$ ./autogen.sh && ./configure
```
I tried 
```
make
```
This fails with error:

```In file included from src/stream_builder.h:19:0,
                 from src/stream_builder.cpp:21:
src/stream_manager.h:20:32: fatal error: avahi-common/watch.h: No such file or directory
```

So unless ```make``` in bare format is not expected, avahi is required.

anselmolsm, 2018-02-26:
d2648cda5cc0ed6b676a2f65999ff9e650e0a58b made avahi dependency enabled by default. 
Avahi is currently installed by the [CI](https://github.com/intel/camera-streaming-daemon/blob/master/.travis.yml) as a dependency. Docs out of sync?

hamishwillee, 2018-02-26:
> Docs out of sync?

@anselmolsm According to @lbegani this should be an "optional" dependency. That makes sense since you actually have a config option `--enable-avahi` - an you can't have a mandatory dependency as an option :-)

So no, not a docs problem, but a code problem.

anselmolsm, 2018-02-26:
@hamishwillee, indeed, logs say e61e2f91 made avahi optional.

lbegani, 2018-02-27:
@anselmolsm @hamishwillee 
We have following compile time flags currently
1. enable-aero : Enables aero specific bottom camera
2. enable-avahi : Enables avahi to advertise rtsp streams
3. enable-mavlink : Enables mavlink support
4. enable-gazebo : Enables gazebo camera 
Reason to provide these configurations are to allow users to pick and choose the combination that fits their requirements. For Intel Aero, we will enable the flags aero, avahi and mavlink.
One of the issues in providing multiple compile time configuration is confusion. Hopefully it can be addressed by proper documentation. Nevertheless, I am OK in making changes based on suggestions to make it better.
Regarding the build error, its getting fixed.

hamishwillee, 2018-02-27:
@lbegani That all sounds fine to me https://github.com/intel/camera-streaming-daemon/issues/115#issuecomment-368747825

Is there an `enable-realsense` options? I ask because it is the "missing" option above from the report at the end of config:
```
>    RealSense support:   no
    MAVLink support:     yes
    AVAHI support:       yes
    Intel Aero support:  yes
    Gazebo support:      no
```
If not can that option ever be anything other than no? If it can be, how does it get enabled?

lbegani, 2018-02-27:
@hamishwillee If you have installed realsense libraries, Realsense support will turn on automatically. There is no explicit option that needs to be given to enable it. **This is missing in documentation**. Can you try installing following packages and see if you get `RealSense support:   yes`
`sudo apt-get install librealsense2-dev`
`sudo apt-get install librealsense2-dbg`

hamishwillee, 2018-02-27:
(as sent to you via slack):

So I followed https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages to get the binaries:
```
sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
sudo apt-get update
sudo apt-get install librealsense2-dev
sudo apt-get install librealsense2-dbg
reboot
```

Then I ran:
```
./autogen.sh && ./configure --enable-aero --enable-mavlink
```
Unfortunately I still see:
```
    RealSense support:   no
```

Ubuntu is giving me system error warnings, so it is possible that there is something else going on, but I'd still like your help.

anselmolsm, 2018-02-27:
librealsense and librealsense2 are not compatible. CSD only supports librealsense, right?

hamishwillee, 2018-02-28:
@anselmolsm @lbegani 

> librealsense and librealsense2 are not compatible. CSD only supports librealsense, right?

Yes, that was the problem. Works when I follow these instructions: https://software.intel.com/sites/products/realsense/intro/getting_started.html

I can proceed with the docs thanks. 

BUT, is it a bug that the we're not compatible with the new SDK? (ie should we be)? If so, new Issue?

anselmolsm, 2018-02-28:
> BUT, is it a bug that the we're not compatible with the new SDK? (ie should we be)? If so, new Issue?

It's a task for the backlog.

lbegani, 2018-02-28:
I tried in a new system and this is what needs to be installed to be able to compile with realsense support
`sudo apt-get install librealsense-dev`
However to run with realsense, we will need all the packages (Intel Aero has them pre-installed).
Can we test CSD on ubuntu with realsense camera?
I haven't tried it but I guess it should work. I used realsense camera with Ubuntu (installed librealsense2) for another project but not for CSD.

hamishwillee, 2018-02-28:
Re https://github.com/intel/camera-streaming-daemon/issues/115#issuecomment-369109326

This is what I used (includes `librealsense-dev`)
```
echo 'deb "http://realsense-alm-public.s3.amazonaws.com/apt-repo" xenial main' | sudo tee /etc/apt/sources.list.d/realsense-latest.list
sudo apt-key adv --keyserver keys.gnupg.net --recv-key D6FB2970 
sudo apt update -y
sudo apt install librealsense-object-recognition-dev librealsense-persontracking-dev librealsense-slam-dev -y
```
It would be useful to differentiate what you need for compiling vs what you need for running on Ubuntu.

PR #123 (2018-03-02): Fix for build error without avahi and miscellaneous warnings
Fixes #115  and #120
closed 2018-03-02 by lbegani