Dronecode/camera-manager #116: Build failed for aero

hamishwillee, 2018-02-26:
Running README instructions for Intel Aero variant failed.

1. Instructions were:
```
sudo apt-get install gstreamer-1.0
sudo apt-get install libgstreamer-plugins-base1.0-dev
sudo apt-get install libgstrtspserver-1.0-dev
sudo pip2 -q install -U future

mkdir github
git clone https://github.com/intel/camera-streaming-daemon.git
cd camera-streaming-daemo
git submodule update --init --recursive

$ ./autogen.sh && ./configure
$ sudo apt-get install libavahi-client-dev libavahi-core-dev libavahi-glib-dev
$ ./autogen.sh && ./configure --enable-aero --enable-mavlink --enable-avahi
```

I additionally installed git and autoreconf

2. To build I just called `make` (there is no specific instruction in README at this point, but it makes sense). 
Build failed as shown below:
```sh
 CXX      src/CameraComponent.o
  CXX      src/CameraServer.o
  CXX      src/CameraDeviceV4l2.o
src/CameraDeviceV4l2.cpp: In member function ‘virtual int CameraDeviceV4l2::resetParams(CameraParameters&)’:
src/CameraDeviceV4l2.cpp:153:13: warning: declaration of ‘ret’ shadows a previous local [-Wshadow]
         int ret = v4l2_set_control(fd, queryctrl.id, queryctrl.default_value);
             ^
src/CameraDeviceV4l2.cpp:135:9: note: shadowed declaration is here
     int ret = 0;
         ^
src/CameraDeviceV4l2.cpp:153:13: warning: unused variable ‘ret’ [-Wunused-variable]
         int ret = v4l2_set_control(fd, queryctrl.id, queryctrl.default_value);
             ^
  CXX      src/ImageCaptureGst.o
  CXX      src/VideoStreamUdp.o
  CXX      src/stream_builder_aero_bottom.o
  CXX      src/stream_aero_bottom.o
  CXX      src/avahi_publisher.o
  CXX      src/mavlink_server.o
src/mavlink_server.cpp:69:23: warning: missing terminating " character
             log_error("Invalid System ID for MAVLink communication (%d).Waiting
                       ^
src/mavlink_server.cpp:70:21: warning: missing terminating " character
              Vehicle",opt.sysid);
                     ^
src/mavlink_server.cpp:747:0: error: unterminated argument list invoking macro "log_error"
 }
 ^
src/mavlink_server.cpp: In constructor ‘MavlinkServer::MavlinkServer(ConfFile&, std::vector<std::unique_ptr<Stream> >&, RTSPServer&)’:
src/mavlink_server.cpp:69:13: error: ‘log_error’ was not declared in this scope
             log_error("Invalid System ID for MAVLink communication (%d).Waiting
             ^
src/mavlink_server.cpp:69:13: error: expected ‘}’ at end of input
src/mavlink_server.cpp:69:13: error: expected ‘}’ at end of input
Makefile:994: recipe for target 'src/mavlink_server.o' failed
make[1]: *** [src/mavlink_server.o] Error 1
Makefile:707: recipe for target 'all' failed
make: *** [all] Error 2
ubuntu@ubuntu:~/github/camera
```

hamishwillee, 2018-02-26:
Same error when configure/make for Ubuntu: https://github.com/intel/camera-streaming-daemon#webcam-ubuntu

anselmolsm, 2018-02-26:
Buld fixed by 2730c7faa2d0f6220ad80bead6794a94d793b7eb. Thanks.

closed 2018-02-26 by anselmolsm

hamishwillee, 2018-02-26:
Awesome. How was this missed by CI?

anselmolsm, 2018-02-26:
The CI currently builds only using the default config flags, thus --enable-aero, --enable-mavlink are not tested. See #117.

hamishwillee, 2018-02-27:
Agree this is fixed. There are still build warnings. Should I create a separate issue for those?
```
 CXX      src/CameraDeviceV4l2.o
src/CameraDeviceV4l2.cpp: In member function ‘virtual int CameraDeviceV4l2::resetParams(CameraParameters&)’:
src/CameraDeviceV4l2.cpp:153:13: warning: declaration of ‘ret’ shadows a previous local [-Wshadow]
         int ret = v4l2_set_control(fd, queryctrl.id, queryctrl.default_value);
             ^
src/CameraDeviceV4l2.cpp:135:9: note: shadowed declaration is here
     int ret = 0;
         ^
src/CameraDeviceV4l2.cpp:153:13: warning: unused variable ‘ret’ [-Wunused-variable]
         int ret = v4l2_set_control(fd, queryctrl.id, queryctrl.default_value);
             ^
  CXX      src/ImageCaptureGst.o
  CXX      src/VideoStreamUdp.o
  CXX      src/mavlink_server.o
src/mavlink_server.cpp: In member function ‘void MavlinkServer::_handle_request_camera_information(const sockaddr_in&, mavlink_command_long_t&)’:
src/mavlink_server.cpp:119:23: warning: comparing floating point with == or != is unsafe [-Wfloat-equal]
     if (cmd.param1 != 1) {
                       ^
src/mavlink_server.cpp: In member function ‘void MavlinkServer::_handle_request_camera_settings(const sockaddr_in&, mavlink_command_long_t&)’:
src/mavlink_server.cpp:151:23: warning: comparing floating point with == or != is unsafe [-Wfloat-equal]
     if (cmd.param1 != 1) {
                       ^
src/mavlink_server.cpp: In member function ‘void MavlinkServer::_handle_request_storage_information(const sockaddr_in&, mavlink_command_long_t&)’:
src/mavlink_server.cpp:182:23: warning: comparing floating point with == or != is unsafe [-Wfloat-equal]
     if (cmd.param2 != 1) {
                       ^
src/mavlink_server.cpp: In member function ‘void MavlinkServer::_handle_request_camera_capture_status(const sockaddr_in&, mavlink_command_long_t&)’:
src/mavlink_server.cpp:289:23: warning: comparing floating point with == or != is unsafe [-Wfloat-equal]
     if (cmd.param1 != 1) {
                       ^
```

lbegani, 2018-02-27:
`src/CameraDeviceV4l2.cpp:153:13: warning: declaration of ‘ret’ shadows a previous local [-Wshadow]`
This is getting fixed , ignore the rest

anselmolsm, 2018-02-27:
The rest should not be ignored. The warning is clear.
I know that float value comes from MAVLink messages. Take a look how other projects using MAVLink deal with such params (PX4, for example).

hamishwillee, 2018-02-27:
I agree @anselmolsm - warnings IMO should either be fixed or supressed. Have created: https://github.com/intel/camera-streaming-daemon/issues/120