Dronecode/camera-manager #118: Coverty build status not displaying on README

hamishwillee, 2018-02-27:
[image]

lbegani, 2018-02-27:
I remember having seen it. Could this be because of 
`Planned Maintenance
This service is temporarily unavailable.
Please try again later. We apologize for any inconvenience.`

hamishwillee, 2018-02-27:
Good point. Lets see if this comes back in a day or so.

hamishwillee, 2018-03-04:
When I click the link on any browser I get a "Site not secure message". That will be the cause of this problem. Is there a "secure URL"?

lbegani, 2018-03-05:
@anselmolsm I remember having seen it working. Is this due to migration?

anselmolsm, 2018-03-05:
"Planned Maintenance - This service is temporarily unavailable." on their side.
"site not secure" = Their certificate might have expired or has been revoked.

The only thing under our control is the badge in README file (and the option would be to remove that for a while).

lbegani, 2018-03-06:
@anselmolsm I would suggest we keep it, if there is a hope that it will come back soon

anselmolsm, 2018-03-16:
It was announced the service is back online.

closed 2018-03-16 by anselmolsm