Dronecode/camera-manager #122: Make have_realsense a flag rather than binary autodetect

hamishwillee, 2018-02-28:
`--have_realsense` - would be more consistent if this was a flag rather than auto-detection of binaries. Can we change it? The system should still fail gracefully if the binaries are not present.

Consistency is important because it simplifies the instructions and makes everything work in a way that people come to expect. 
This is the only case where we auto-check for binaries. For example, in Gazebo we don't.

lbegani, 2018-02-28:
What would be more convenient to the users - 
1. Have auto-detection for both gazebo and realsense. Have flags for other options (like mavlink etc)
2. Have explicit flags for everyone, as you said
3. Have no compile time flag at all, compile everything available, enable/disable features runtime based on input conf file (little far-stretched, not sure if there can be issues with this approach)

hamishwillee, 2018-02-28:
Option2.
- '1' is a different mix of inconsistent, but still inconsistent.
- '3' is essentially the same as 2, except that you have a file. If you were going to be doing config every build cycle then this would be a good option, but we're not, so it is just annoying.

anselmolsm, 2018-02-28:
It's due to historic reasons. A config flag does not exist because librealsense was required, not optional. I mean, in the past it wasn't possible to build CSD without that library.

Option 2 makes sense.

lbegani, 2018-03-01:
We will go with option 2 then. Should I add a flag as `--enable-realsense` instead of `--have_realsense` for consistency. If the flag is enabled during compile time AND realsense dev libs are installed, then realsense support will be added in CSD by compiling related code.

hamishwillee, 2018-03-01:
Yes, use `--enable-realsense`. 

> If the flag is enabled during compile time AND realsense dev libs are installed, then realsense support will be added in CSD by compiling related code.

Yes. If `--enable-realsense` is specified and the realsense dev libs are not present the build should fail gracefully in the configure step.

lbegani, 2018-03-06:
Done - https://github.com/intel/camera-streaming-daemon/pull/128

closed 2018-03-06 by lbegani