Dronecode/camera-manager #126: Proposing a better name

shakthi-prashanth-m, 2018-03-01:
The name _Camera Streaming Daemon_ sounds like it does only streaming job. It does many thing other than just streaming!
And the word _Daemon_ comes from Unix/Linux jargon (process running in the background) and its not a high level name, because a daemon not necessarily a service provider. What does it convey with it ?

I think  **Camera service** or **Camera server** could be simple and precise names matching its functionality.

Makes sense ?

anselmolsm, 2018-03-01:
Historic reasons. But final decision about the name is not ours.

closed 2018-03-01 by anselmolsm

lbegani, 2018-03-02:
I prefer to call it `Camera Server` in most of my discussions. However changing it in the repo seems out of our scope.