Dronecode/camera-manager #127: Better logging if  config path not found?

hamishwillee, 2018-03-02:
First thing I tried (without thinking) on ubuntu was: `./csd`. This failed with error below because `/etc/csd/main.conf` is not present unless you put it there. 

[image]

I'm wondering if we can make this a tiny bit better for developers:
1. Check for existence of the config file in CSD (default or otherwise) and fail with better message - e.g.:
   > Specified conf file not found (`whatever file it is`). Use `--conf` flag to specify configuration file.
2. Even if we don't do this, would be good to get rid of the other warnings about missing location etc, (as these are spurious/just because of the missing file.
3. Perhaps have a second default file `./samples/files/ubuntu.conf`. Not sure about this, since someone might choose to run on Arch Linux or whatever.

hamishwillee, 2018-03-06:
@lbegani FYI

lbegani, 2018-03-28:
More than logging, it is about handling of no or invalid conf file. Currently, if there is no conf file or dir with conf files, that gets parsed successfully, the ConfFile object would be empty.  This empty object is getting passed around. Also, there is no public method to check if the ConfFile obj is invalid. I am thinking of adding changes where ConfFile with no entries will not be passed. This will require making the usage of the ConfFile obj optional in CameraServer, MavlinkServer and RTSPServer. Keeping the issue open till done.