Dronecode/camera-manager #129: Deploying on Aero - confirm steps

hamishwillee, 2018-03-05:
@lbegani For you :-)

Wiki has [Deploying on Aero](https://github.com/intel/camera-streaming-daemon/wiki/Deploying-on-Aero).  
1. Is it still necessary to rebuild the Aero image from master branch? If not, can you just use the image already on the vehicle - or is there some other image you should use?
2.  Second step is to "boot Aero board and set up an ssh connection with your computer, and copy CSD to Aero board:
   ```
   $ scp camera-streaming-daemon 192.168.1.1:
   ```
   - I think that this is out of date - the executable was formerly called *camera-streaming-daemon* and is now called *csd*
   - I know/imagine we need to copy three things - csd executable, csd.service and the aero.conf file. How can we do this and put them in the right place(s)?
3. Next step is to run CSD, which the wiki article does manually. It is correct that instead you would just boot the board and it should start CSD automatically via systemd?

lbegani, 2018-03-06:
Yes, the wiki is outdated. 

1.  No, released aero image(1.6.1) now has csd integrated
2.  Not required to copy csd bin anymore
3.  Yes, boot the board, csd is there running by default.

hamishwillee, 2018-03-06:
Thanks @lbegani 
But what if I want to update CSD with a later version? Do I rebuild the image with my updated CSD or copy it or what?

Or is this not a real use case and the doc should just be deleted (it "feels" real)!

lbegani, 2018-03-06:
For latest CSD, we need to compile and copy the csd binary.
The doc was relevant earlier as csd was not part of the aero image. We can still keep the updated steps with a note.

hamishwillee, 2018-03-06:
Ah, so you just copy the csd binary - but you don't need the other files because they won't have changed (ie csd.system and the config file)?

lbegani, 2018-03-06:
Yes, if other files have not changed. However its possible that the conf file is changed by the user, then the conf file will need to be copied as well. I do not see a reason why service file will need to be changed.

hamishwillee, 2018-03-06:
Cool. So my concern is how you get the files to the right place. IF you do
```
$ scp csd  192.168.1.1:
```
Will that copy csd to the right place in the folder structure/image? Where should it go so that it is picked up by systemd - and how do you get it there?

hamishwillee, 2018-03-06:
@lbegani And similarly, 
> Yes, if other files have not changed. However its possible that the conf file is changed by the user, then the conf file will need to be copied as well. I do not see a reason why service file will need to be changed.

Assuming this, again, how do I copy this file onto the device in the "right" place.

lbegani, 2018-03-07:
To copy csd binary
`$ scp csd  uname@ip-addr:/usr/bin/`
To copy csd conf file
`$ scp myconfig.conf  uname@ip-addr:/etc/csd/main.conf`

For Yocto, `uname = root`,` ip-addr=192.168.7.2` (check using ifconfig)
For Ubuntu, `uname = <user-defined>`, `ip-addr=192.168.7.2 `(check using ifconfig)

hamishwillee, 2018-03-21:
Thanks @lbegani - Closing now.

closed 2018-03-21 by hamishwillee