Dronecode/camera-manager #132: camera-sample-mavlink-client fails on Ubuntu

hamishwillee, 2018-03-06:
1. The CSD will need to be run with a config that matches the platform. For [ubuntu.conf](https://github.com/intel/camera-streaming-daemon/blob/master/samples/files/ubuntu.conf) the mavlink config is
   ```
   port=24570
   system_id=1
   ```
   - However the example is hard coded to connect to port 14550 and will not detect CSD for this configuration.
   - Why is this port and system id used?
1. If the port is changed in config to 14550 the example runs, but it **does not detect any streams**:
   ```
   Open UDP [5]
   Bind UDP [5] 0.0.0.0:14550
   Camera Daemon found: sysid: 1

   No streams found.
   Please make your selection (type stream number):
   ```

So I suggest that we update sample to take an argument for the port (with default 14550) but still need debugging on why streams not found.

hamishwillee, 2018-03-06:
Can we also confirm that the purpose of this example is to "Test MAVLink publishing of available RTSP streams" ? ie as an alternative tor Avahi

hamishwillee, 2018-03-13:
This still fails with changes in https://github.com/intel/camera-streaming-daemon/pull/138/

lbegani, 2018-03-13:
@hamishwillee Lot of things have changed in the mavlink protocol since the time this was written. The idea earlier was to test mavlink video streaming message but now I would like to change it to be a mavlink client to do what QGC does, detect hearbeat, query camera caps etc. For now , we have just moved the files around, no work on making the test app is done yet.  We will make it functional and keep you posted.

hamishwillee, 2018-03-13:
Thanks @lbegani OK. I've done https://github.com/Dronecode/csd_docs/pull/5 to just track the files moving. There are a few questions there :-)