Dronecode/camera-manager #139: Aero conf URL broken

hamishwillee, 2018-03-08:
See https://github.com/intel/camera-streaming-daemon/blob/master/samples/files/aero.conf#L16
Missing equals sign and "h"
```
video13ttp://127.0.0.1/camera-def-R200rgb.xml
```
This will be un-broken when #136 merges.

@lbegani Question - does aero really serve to http://127.0.0.1/? In a "real" vehicle the URL will be supplied in the camera protocol. Is this accessible to QGC?

- Or to put it another way, is it likely/realistic that in most cases a vehicle will host its own camera definition files? Or is it more likely these will be on the net somewhere?

lbegani, 2018-03-09:
I think there is a mismatch between aero.conf and the conf file actually used with Intel Aero - https://github.com/intel-aero/meta-intel-aero/blob/master/recipes-support/camera-streaming-daemon/files/main.conf
I will update it after few tests, keeping the issue open till then.

hamishwillee, 2018-03-21:
@lbegani Great - can you please assign this issue to you then? Also, how is this coming along?

hamishwillee, 2018-03-21:
This appears to have been fixed

closed 2018-03-21 by hamishwillee