Dronecode/camera-manager #142: Support 6 cameras

hamishwillee, 2018-03-09:
Currently the CSD supports only 5 cameras even though MAVLink allocates 6 component IDs. The reason for this is that PX4/SITL reserves the first camera component id - 100 (for the basic plugin). 

The suggestion is to allocate component IDs from 100 by default, but enable from 101 (as currently) iff `--enable-gazebo` is set.

@lbegani As discussed.

lbegani, 2018-03-28:
While testing CSD (with or without --enable-gazebo) with simulator (PX4-SITL+Gazebo) , comp ID 100 cannot be used as this will conflict with the gazebo camera component with ID 100. Only option left is to use ID 100 for --enable-aero and Aero doesn't need that many IDs.

hamishwillee, 2018-03-28:
Interesting. I see two other options:
1. Config file - set id of first mavlink camera (preferred)
2. An enable flag

lbegani, 2018-03-29:
Change is trivial by will add complexity (for the user). I feel the compile time flags and configurations in conf file should be minimal. 
Secondly, there seems to be an issue with QGC in addressing comp with IDs other than 100. Need a fix there as well.
Am still not decided on this.

hamishwillee, 2018-03-29:
To me this depends mostly on whether anyone will really ever want more than 5 cameras. If they will, then "someone" will need a workaround - either more ids in MAVLink or a fix as proposed.

I don't have any problem with the "complexity" of having the config flag, with default of the second component ID. The risk for me here is that the user does not understand the circumstances in which they can change the value.

lbegani, 2018-04-23:
@hamishwillee Can we close this issue?

hamishwillee, 2018-04-24:
Yes. Thanks very much.

closed 2018-04-24 by hamishwillee