Dronecode/camera-manager #144: ubuntu.conf needs to be updated to use SimpleHTTPServer port

hamishwillee, 2018-03-14:
As discussed, it is much easier to use SImpleHTTPServer to serve Camera Definition Files than Apache (this is, after all, what Aero does). 
SimpleHTTPServer serves to a particular port - e.g. 8000.

The requirement is to update the ubuntu.conf so that it uses such a port (e.g. 8000 or 8080 or similar) so that we can set add  instructions using SimpleHTTPServer in the guide.

@lbegani I know you know about this. I'm adding the issue so I can track it.

lbegani, 2018-03-14:
Thanks @hamishwillee . I will do it as a part of PR - https://github.com/intel/camera-streaming-daemon/pull/140
I think both gazebo.conf and ubuntu.conf  will need to be updated

hamishwillee, 2018-03-21:
This has been fixed

closed 2018-03-21 by hamishwillee