Dronecode/camera-manager #149: Handling of system ID: default, from conf file or from autopilot

lbegani, 2018-03-27:
@hamishwillee @SaliniVenate 
For System ID, this could be a strategy to follow for CSD -

1. Read from the conf file (-1 : read from network, >=0 : use as it is )

    - If conf file is not there, use some default value (1)

2. Read from the network (if system ID is set to -1 in conf file)

    - Start sending camera heartbeat with default system ID (1)
    - On receiving autopilot heartbeat, send camera heartbeat with updated system ID.
    - On receiving heartbeats from multiple autopilots, stick to the first one

hamishwillee, 2018-03-27:
Sounds good to me. In 90% of cases this will be connected to autopilot with sys id 1 and this will just work. 

The only problem I see is the case of multiple autopilots (small fraction of time) - should we somehow report/log multiple autopilots on the network so users can know this is a possible problem?

lbegani, 2018-04-23:
@hamishwillee Can we close this issue?

hamishwillee, 2018-04-23:
Nearly. Can you update the defaults in the sample file - https://github.com/Dronecode/camera-streaming-daemon/blob/master/samples/config/config.sample#L20 e.g. system id default is now effectively 1.