Dronecode/camera-manager #159: Rename Camera Streaming Daemon

anitha-suresh, 2018-04-11:
The discussion to rename the "camera streaming daemon", has been happening in pockets.

One of the suggestions, was to rename it as : "Dronecode Camera Server".

Feel free to share your opinion here.

hamishwillee, 2018-04-11:
I like both above suggestions. Any others?
- Dronecode Linux Camera Server
- Dronecode Camera Daemon
- Dronecode Camera Manager
- Dronecode Camera Service

@mrpollo @anitha-suresh Who else should be part of this discussion?

hamishwillee, 2018-04-13:
@mrpollo @anitha-suresh Ping https://github.com/Dronecode/camera-streaming-daemon/issues/159#issuecomment-380619216

I'm sure you're all aware that if you want me to work on this then we're on a timer :-0

hamishwillee, 2018-04-13:
@anitha-suresh FYI, I had a chat to @mrpollo about this. He is leaning towards "Dronecode Camera Manager". So:
- Github: dronecode/camera-manager 
- main entry point: http://dronecode.org/camera-manager
- Docs **probably**: http://docs.dronecode.org containing BOTH SDK and Camera Manager 
That last one still a bit under discussion.

Anyway, not sure what it takes to make this **agreed** - perhaps a vote in the dev meeting?

mrpollo, 2018-04-13:
@anitha-suresh I think we should use Dronecode Camera Manager, the name is clear and to the point.

@LorenzMeier What do you think of the name?

lbegani, 2018-04-14:
@hamishwillee @mrpollo 
My preference would be to name CSD as `camera-server` with the following reasoning-

- It provides camera related services like video streaming, video capture, image capture and camera control
- The services are provided to MAVLink client, so it more fits into client-server model.

We can choose to have or not have `Linux` term attached. It has been only tested on linux based platform but porting it to other platforms should not be difficult given that the code is mostly standard C++11 (and gstreamer).

lbegani, 2018-04-16:
@hamishwillee 
I think we can choose to fit the software into client-server paradigm, as from outside, it is servicing requests from remote user and is running all the time. RTSP streaming is certainly client-server (CSD is running GstRTSPServer). 

I am OK with anything that sounds good to majority.

hamishwillee, 2018-04-18:
@lbegani "Server" doesn't fully capture the backend integration, while "manager" doesn't fully capture the idea of providing services. Both are OK though, and work better than CSD.

Upshot, we just need *someone* to make a final decision. Soon.

anitha-suresh, 2018-04-18:
I prefer Dronecode 'Camera Manager' to have the broader scope - manage/include anything and everything about camera.

hamishwillee, 2018-04-19:
And the final decision from the dev meeting  is .... [drum roll] ... ?

hamishwillee, 2018-04-23:
@anitha-suresh @mrpollo I assume the lack of response means that you both heartily support the camera manager name?

@lbegani let's plan this. When do you think you might be able to move the repo to the suggested location?

@mrpollo I have a problem with having both SDK and Camera Manager at the same URL. The problem is that the SDK/Dronecore docs versioned and in any one docset you can only have one version. What that means is that I can only have docs for these together if they share exactly the same release schedule and versioning.
It is probably easier to separate the docs, at least at this point. So perhaps http://sdk.dronecode.org and camera-manager.dronecode.org ?

mrpollo, 2018-04-23:
I think we should use "Camera Manager".

sdk.dronecode.org and camera-manager.dronecode.org both work, lets use them.

hamishwillee, 2018-04-24:
Great. That is what we are going to do then. Any change of mind after this point is too late.

lbegani, 2018-04-24:
> @lbegani let's plan this. When do you think you might be able to move the repo to the suggested location?

@anitha-suresh @mrpollo As you guys were involved in the movement earlier, can you guide with the steps?

lbegani, 2018-04-24:
I have renamed the repo from `camera-streaming-daemon` to `camera-manager`

hamishwillee, 2018-04-24:
And docs are all updated. This can be closed.

@mrpollo I do need that DNS redirect so I can assign the URL for the docs.

closed 2018-05-17 by lbegani