Dronecode/camera-manager #161: Support configuration of Image Capture and Video Capture from GCS

lbegani, 2018-04-18:
Support export of Image Capture and Video Capture settings to GCS through camera definition file. At present, CSD only handles parameters to configure camera device. The parameters for image capture or video capture need to be supported by adding handlers in respective classes.

eeroniemi, 2018-07-13:
This would be really useful feature. I'm not totally sure if plan is that these values could be modified also on `[vidcap]` section on configuration file - that would be nice.

I'm working on a project where we need to stream video from Intel RealSense r200 over 4G/LTE connection - with default settings the video bandwidth usage is about 100 Mbit/s - way too much to stream over mobile network. When I modified gstreamer width and height parameters directly on the source code (these two lines: https://github.com/Dronecode/camera-manager/blob/master/src/gstreamer_pipeline_builder.cpp#L61-L62) with different values I was able to get good results.

Options I tried:

- Default values, no change to code. Bandwidth usage: 100 Mbit/s, video resolution 1920 x 1080.
- width = 960, height = 540. Bandwidth usage: 5 Mbit/s. <- totally usable over LTE
- width = 1920, height = 1080. Bandwidth usage: 33 Mbit/s. <- this is something I don't really understand. Just by explicitly setting these values to original video resolution the bandwidth usage dropped this dramatically?

lbegani, 2018-07-13:
@eeroniemi For RTSP streaming, you can specify the width and height in the URL, change in source code is not required. For example - `rtsp://192.168.8.1:8554/video13?width=960&height=540`
Not sure about the drop in BW usage issue, but bitrate of the output video stream is also dependent on the scene and movements in it.

eeroniemi, 2018-07-13:
@lbegani Thank you for the tip - setting the stream width and height on the URL seems to work without any code changes.

For the bandwidth usage, I had the camera at the same location on each of the tests with nothing dramatically different happening on the video. Not a big issue now when I know about it, but just interesting insight.