Dronecode/camera-manager #182: Frame rate/bit rate/width/height params for streaming

seanrowens, 2018-09-05:
The documentation on config files is a tad unclear, so maybe I'm doing this wrong.  I tried to set frame rate to a 5 and image width/height to 640x480, but when I stream the video I'm still seeing 1920x1080 images at 30FPS.  Someone on slack said those parameters only apply to vidcap and not to streaming, and suggested that I file this issue.

hamishwillee, 2018-09-05:
I suspect the vidcap is for video saving, not video streaming. I don't believe there are parameters to control the RTSP streaming but @lbegani will know for sure.

Lalit, when this is confirmed, I'll tidy the docs for vidcap to make this clear.

hamishwillee, 2018-09-10:
Thanks @lbegani . Docs fixed in https://github.com/Dronecode/camera-manager-docs/commit/6b0167934a16c7ead9d7e069db43f6acfc007723

Let's leave this open to track vidstream - I'll rename to make it clear.