Dronecode/camera-manager #185: Is there any tips for reduce the video stream latency?

Huibean, 2018-11-19:
the latency is about 2s, any tips for improving this?
[image]

network status
```
➜  ~ ping 192.168.1.123
PING 192.168.1.123 (192.168.1.123): 56 data bytes
64 bytes from 192.168.1.123: icmp_seq=0 ttl=64 time=96.275 ms
64 bytes from 192.168.1.123: icmp_seq=1 ttl=64 time=18.170 ms
64 bytes from 192.168.1.123: icmp_seq=2 ttl=64 time=36.746 ms
64 bytes from 192.168.1.123: icmp_seq=3 ttl=64 time=58.287 ms
64 bytes from 192.168.1.123: icmp_seq=4 ttl=64 time=79.758 ms
64 bytes from 192.168.1.123: icmp_seq=5 ttl=64 time=97.455 ms
64 bytes from 192.168.1.123: icmp_seq=6 ttl=64 time=19.504 ms
64 bytes from 192.168.1.123: icmp_seq=7 ttl=64 time=38.604 ms
64 bytes from 192.168.1.123: icmp_seq=8 ttl=64 time=63.328 ms
```

lbegani, 2018-11-20:
Can you reduce the resolution and check. I guess this is mostly due to network congestion.

MatejFranceskin, 2019-02-18:
With the upper PR you can have the control over video streaming parameters. In the file samples/config/ubuntu.conf you can see an example that creates RTSP gstreamer pipeline with tune=zerolatency option that reduces latency to minimum.

[rtsp]
pipeline=v4l2src device=/dev/video0 ! videoconvert ! video/x-raw, format=I420 ! x264enc speed-preset=ultrafast tune=zerolatency ! rtph264pay name=pay0

I think this should solve your problem.

closed 2019-03-07 by MatejFranceskin