Dronecode/camera-manager #193: please help to setup RS r200 in ubuntu 18.04

FPSychotic, 2019-03-20:
HI , followed all possible configurations that I could imagin to make work Realsense r200 in ubuntu 18.04 and GPU Intel i960 driver i Think
the camera start, as the IR emiter is turn on, the streamings are made but imposible get image in my IP or 127.0.0.1
for ubuntu when I made the dcm should I config for Aero, Aero+realsense or only realsense. should I start the server for xml and aiim the uri to rs-rgb, or uvc?
is mandatory have working the mavlink link?
i get image from a webcam with no problem but I cant with r200, maybe the gstreamer config is bad for 18.04?

MatejFranceskin, 2019-03-20:
I don't have the hardware to try it but maybe I can help you. First start dcm and post the output so that we can see how the camera was detected.

FPSychotic, 2019-03-30:
**I have ubuntu 18.04, realsense r200, usb camera uvc,and gazebo.
I make this steps and I get this outs**
 $cd /home/ihexa/camera-manager/samples/def
Serving HTTP on 0.0.0.0 port 8000 ...

$./autogen.sh && ./configure --enable-mavlink --enable-realsense --enable-avahi --enable-gazebo
...
    RealSense support:   yes
    MAVLink support:     yes
    AVAHI support:       yes
    Intel Aero support:  no
    Gazebo support:      yes
    Custom support:      no

$make
...
  CXX      src/main.o
  CXXLD    dcm
  GEN      dronecode-camera-manager.service

***I run mavros via UART ttys04 921600bds (connect to FCU chibiOS via TL2 port)  and it serve with its proxy url_gcs (in launch file)in port 14450 to mavlink-router which will distribute to internal gcs in companin gcs port 14551, gcs in netbook port 14552. The conexion is all by an access point made with my phone (192.168.43.1),but with home router I get same results.***
MAVROS LAUNCH
<arg name="fcu_url" default="/dev/ttyS4:921600" />
        <arg name="gcs_url" default="udp://@127.0.0.1:14550" />
	<arg name="tgt_system" default="1" />
	<arg name="tgt_component" default="1" />
MAVROS OUT
...
[ INFO] [1553952945.742400657]: FCU URL: /dev/ttyS4:921600
[ INFO] [1553952945.749554984]: serial0: device: /dev/ttyS4 @ 921600 bps
[ INFO] [1553952945.761497870]: GCS URL: udp://@127.0.0.1:14550
[ INFO] [1553952945.761992498]: udp1: Bind address: 0.0.0.0:14555
[ INFO] [1553952945.762191889]: udp1: Remote address: 127.0.0.1:14550
...
 WARN] [1553952946.853787048]: IMU: linear acceleration on RAW_IMU known on APM only.
[ WARN] [1553952946.853911680]: IMU: ~imu/data_raw stores unscaled raw acceleration report.
[ WARN] [1553952946.855662164]: GP: No GPS fix
[ INFO] [1553952947.214916150]: CON: Got HEARTBEAT, connected. FCU: ArduPilot
[ INFO] [1553952947.342551352]: RC_CHANNELS message detected!
[ INFO] [1553952948.231186996]: VER: 1.1: Capabilities         0x00000000000031cf
[ INFO] [1553952948.231300266]: VER: 1.1: Flight software:     03050000 ( b450355)
[ INFO] [1553952948.231364088]: VER: 1.1: Middleware software: 00000000 (        )
[ INFO] [1553952948.231433222]: VER: 1.1: OS software:         00000000 ( 1052fcc)
[ INFO] [1553952948.231499094]: VER: 1.1: Board hardware:      00000000
[ INFO] [1553952948.231558366]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1553952948.231622264]: VER: 1.1: UID:                 0000000000000000
[ WARN] [1553952948.231763320]: CMD: Unexpected command 520, result 0
[ INFO] [1553952957.215623927]: HP: requesting home position
[ INFO] [1553952957.219801564]: FCU: ArduRover V3.5.0-dev (b450355f)
[ INFO] [1553952957.221730789]: FCU: ChibiOS: 1052fcc7
[ INFO] [1553952957.224452978]: FCU: mini-pix 0055003C 3136510E 34373935
[ INFO] [1553952962.218387252]: WP: mission received
[ INFO] [1553952967.216354178]: HP: requesting home position

i START MAVLIN-ROUTER
$ mavlink-routerd -e 192.168.43.156:14552 -e 127.0.0.1:14553 -e 127.0.0.1:14551  127.0.0.1:14550
Open UDP [4] 127.0.0.1:14550 *
Open UDP [5] 127.0.0.1:14551  
Open UDP [6] 127.0.0.1:14553  
Open UDP [7] 192.168.43.156:14552  
Open TCP [8] 0.0.0.0:5760 *

**DCM main.cfg as I want realsense r200 rgb camera streaming so I use the aero.cfg file which I adapt to my ls /dev/video* output, my companion IP and port that is serve by mavlink-router 14553**

**MAIN.cfg in /etc/dcm**
[gstreamer]
muxer=rtph264pay
encoder=vaapih264enc
converter=autovideoconvert

[... 302 lines cut ...]
cb_media_configure
cb_unprepared
cb_create_element
host:127.0.0.1,port:8554,abspath:/video0,query:(null)
GST Pipeline: v4l2src device=/dev/video0 ! videoconvert ! video/x-raw, format=I420 ! x264enc speed-preset=ultrafast tune=zerolatency ! rtph264pay name=pay0
cb_media_configure
cb_unprepared
cb_create_element
host:127.0.0.1,port:8554,abspath:/video0,query:(null)
GST Pipeline: v4l2src device=/dev/video0 ! videoconvert ! video/x-raw, format=I420 ! x264enc speed-preset=ultrafast tune=zerolatency ! rtph264pay name=pay0
cb_media_configure
cb_unprepared
cb_create_element
host:127.0.0.1,port:8554,abspath:/video0,query:(null)
GST Pipeline: v4l2src device=/dev/video0 ! videoconvert ! video/x-raw, format=I420 ! x264enc speed-preset=ultrafast tune=zerolatency ! rtph264pay name=pay0
cb_media_configure
cb_create_element
host:127.0.0.1,port:8554,abspath:/video0,query:(null)
GST Pipeline: v4l2src device=/dev/video0 ! videoconvert ! video/x-raw, format=I420 ! x264enc speed-preset=ultrafast tune=zerolatency ! rtph264pay name=pay0
UDP: [12] wrote 21 bytes
UDP: [12] wrote 21 bytes
...
**No video, if I use the uvc.xml and same main.cfg with my usb camera it works in qgc in the companion but not in the computer neither**

thank you!!

MatejFranceskin, 2019-03-30:
Is the graphics card on your computer Intel GPU or something else?

FPSychotic, 2019-03-30:
> Is the graphics card on the computer where dcm is running Intel GPU or something else?

exactly it is an Aaeon UP 4Gb with Intel HD400

MatejFranceskin, 2019-03-31:
In your output I see lines like:
RTSP stream ready at rtsp://:8554/rsir

Did you try to use 
ffplay rtsp://127.0.0.1:8554/rsir
ffplay rtsp://127.0.0.1:8554/rsir2
ffplay rtsp://127.0.0.1:8554/video1
ffplay rtsp://127.0.0.1:8554/video2

instead of:

rtsp://127.0.0.1:8554/video0

ffplay is a command line video player that gives you more output about what is going on.

FPSychotic, 2019-03-31:
No I didn't test them as they are the IR cameras of the realsense, the rgb camera is video0. 
I will take a look with ffplay.
I think I need a Gstreamer pipeline for vaapi codecs as my gpu is compatible with this api. M

MatejFranceskin, 2019-03-31:
This pipeline should work for hadrware accelerated video on Intel HD graphics:
pipeline=v4l2src device=/dev/video1 ! videoconvert ! video/x-raw, format=I420 ! vaapih264enc ! rtph264pay name=pay0

Can you try to run dcm like this, it will generate a lot of output but we might see what is wrong:
GST_DEBUG=3 ./dcm -c path_to_your_config

FPSychotic, 2019-03-31:
Ohh thank you!!! Even if not work is a lot more than I can make my self.
I tried ffplay with DCM with .conf file and without it, didn't worked. I tested with a realsense d435 too, but didnt work either. Maybe DCM works with realsense r200 because Intel aero RTF drone has a FPGA.
I get video with the webcam in ffplay and other clients in 127.0.0.1 but I cannot get image from other computers neither.
Im going to try that pipeline, thank you again

MatejFranceskin, 2019-03-31:
D435 needs librealsense 2.0 and we currently don't support it - support will be added in the future. I thought that R200 is old lib but it might require 2.0

FPSychotic, 2019-03-31:
I have both r200 wirh sdk1 and d435 with sdk2 I just tried because dont lose anything xD.
oMG, now I fall, RTSP ports are tcp or udp, because I opened udp, tcp probably are close

FPSychotic, 2019-03-31:
I have open udp and tcp ports and this is the main.cfg to can at least get visualize the r200s rgb  camera.

[mavlink]
port=14553
broadcast_addr=127.0.0.1
rtsp_server_addr=127.0.0.1
system_id=1

[uri]
video0=http://127.0.0.1:8000/camera-def-rs-rgb.xml

[imgcap]
width=640
height=480
format=2
location=/tmp/

[vidcap]
width=640
height=480
framerate=30
bitrate=1000
encoder=3
format=1
location=/tmp/

[rtsp]
pipeline=v4l2src device=/dev/video0 ! videoconvert ! video/x-raw, format=YUY2 ! vaapih264enc ! rtph264pay name=pay0

ffplay output
.....
enable-frei0r --enable-libopencv --enable-libx264 --enable-shared --enable-version3 --disable-doc --disable-programs --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libtesseract --enable-libvo_amrwbenc
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100
    nan    :  0.000 fd=   0 aq=    0KB vq=    0KB     nan    :  0.000 fd=   0 aq=    0KB vq=    0KB [rtsp @ 0x7fb6fc000b80] method DESCRIBE failed: 503 Service Unavailable
rtsp://127.0.0.1:8554/video0: Server returned 5XX Server Error reply
    nan    :  0.000 fd=   0 aq=    0KB vq=    0KB sq=    0B f=0/0   
ihexa@ihexa-UP-CHT01:~$ ffplay rtsp://127.0.0.1:8554/video0

DCM repeat this output constantly

ull)
GST Pipeline: v4l2src device=/dev/video0 ! videoconvert ! video/x-raw, format=I420 ! vaapih264enc ! rtph264pay name=pay0
host:127.0.0.1,port:8554,abspath:/video0,query:(null)
GST Pipeline: v4l2src device=/dev/video0 ! videoconvert ! video/x-raw, format=I420 ! vaapih264enc ! rtph264pay name=pay0
host:127.0.0.1,port:8554,abspath:/video0,query:(null)
i dont wamt disturb probably Im beginner and Im doing something wrong
if I distract to much we can close the thread

FPSychotic, 2019-03-31:
GST_DEBUG=3 dcm it sent many errors

AVAHI START
host:127.0.0.1,port:8554,abspath:/video0,query:(null)
GST Pipeline: v4l2src device=/dev/video0 ! videoconvert ! video/x-raw, format=I420 ! vaapih264enc ! rtph264pay name=pay0
0:00:15.919976792 12655 0x7ff230005770 WARN                 default v4l2-utils.c:189:gst_v4l2_error:<v4l2src0> error: Device '/dev/video0' is busy
0:00:15.920171858 12655 0x7ff230005770 WARN                 default v4l2-utils.c:192:gst_v4l2_error:<v4l2src0> error: Call to S_FMT failed for YUYV @ 320x962: Device or resource busy
0:00:15.920500168 12655 0x7ff238010850 WARN               rtspmedia rtsp-media.c:2720:default_handle_message: 0x7ff23805a290: got error Device '/dev/video0' is busy (gstv4l2object.c(3709): gst_v4l2_object_set_format_full (): /GstPipeline:media-pipeline/GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
Call to S_FMT failed for YUYV @ 320x962: Device or resource busy)
0:00:15.920521517 12655 0x7ff230005770 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<v4l2src0> error: Internal data stream error.
0:00:15.920729358 12655 0x7ff230005770 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<v4l2src0> error: streaming stopped, reason not-negotiated (-4)
0:00:15.921275170 12655 0x7ff238010850 WARN               rtspmedia rtsp-media.c:2720:default_handle_message: 0x7ff23805a290: got error Internal data stream error. (gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:media-pipeline/GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4))
0:00:15.922492764 12655 0x5614916c91e0 WARN               rtspmedia rtsp-media.c:2989:wait_preroll: failed to preroll pipeline
0:00:15.922659006 12655 0x5614916c91e0 WARN               rtspmedia rtsp-media.c:3293:gst_rtsp_media_prepare: failed to preroll pipeline
0:00:15.927325452 12655 0x5614916c91e0 ERROR             rtspclient rtsp-client.c:1044:find_media: client 0x5614918b2150: can't prepare media
0:00:15.928539521 12655 0x5614916c91e0 ERROR             rtspclient rtsp-client.c:2898:handle_describe_request: client 0x5614918b2150: no media